Livox-SDK/Livox-SDK2 #1: Livox HAP - frame_cnt一直输出0

JIANever404, 2022-08-19:
我们在LivoxViewer2上设置好了Livox HAP，在LivoxViewer2上显示都正常的。
将sdk的demo跑起来之后，点云数据输出的包，在回调里面解析出来frame_cnt一直是0；udp_cnt每一包+1，从0加到65535循环。
应该怎么判断点云的一帧数据结束。
谢谢！
[image]
[image]

asflower, 2023-04-08:
同样的疑问，没发现点云包里面有相关的标记

soul-coder-heap, 2023-06-12:
+1 为啥没人解答

Livox-Infra, 2023-06-12:
hi all, 
对于 HAP lidar，frame_cnt 字段会一直为 0。详见 HAP 通信协议：https://github.com/Livox-SDK/Livox-SDK2/wiki/Livox-SDK-Communication-Protocol-HAP

Livox 通信协议是平台化协议，frame_cnt 是出于兼容考虑而保留在 HAP 点云中

samsaber1998, 2024-04-07:
朋友你好，这个问题我再次遇到了，ufpcnt一直计数到极大值然后清零，无法得知帧的开始和结束是哪个package，你有办法了吗？