Livox-SDK/Livox-SDK2 #4: How to make it run?

Minsu-Kim-KNU, 2022-08-25:
Hi all,

I try to use with sample, connection is right but my HAP does not run like in the windows environment.

How to make it run??

[image]

maskjp, 2022-09-27:
Hi, @Minsu-Kim-KNU,
Are you able to run you HAP now? We met the same problem and have no idea how to fix it?

@Livox-SDK 
Can you help us out?

Minsu-Kim-KNU, 2022-09-28:
@maskjp Hi,

This package is availabe on Ubuntu18.04 and ROS melodic.
I think you have to check out the IP address in the configure file.

zhengsiyusy, 2023-02-23:
The following error occurs when I run the sample in mid-360 Livox-SDK2-master. May I ask how I should successfully run the sample
```
[2023-02-23 14:39:05.414] [console] [info] Livox lidar logger disable.  [parse_cfg_file.cpp] [Parse] [84]
[2023-02-23 14:39:05.414] [console] [info] Data Handler Init Succ.  [data_handler.cpp] [Init] [49]
bind failed
[2023-02-23 14:39:05.414] [console] [error] Create detection socket failed.  [device_manager.cpp] [CreateDetectionChannel] [268]
[2023-02-23 14:39:05.414] [console] [error] Create detection channel failed.  [device_manager.cpp] [CreateChannel] [224]
[2023-02-23 14:39:05.414] [console] [error] Create channel failed.  [device_manager.cpp] [Init] [144]
Failed to init livox lidar sdk.
[ERROR] [1677134345.414607311]: Init lds lidar failed!
```
The running environment is ubuntu18.04.

Livox-Infra, 2023-03-13:
> 

hi, pls check the ips set in "host_net_info" in the config file and those should be set to your host (computer) ip

Victor-Boyd, 2024-04-02:
So for me the problem was the host net ip. It was set to 192.168.1.5 by default but I checked my ip and it was 192.168.1.50. That one digit on the end should fix everything.