Livox-SDK/Livox-SDK2 #7: Running sample livox_lidar_quick_start

changkk, 2022-10-12:
Hi, I am trying to run HAP with the sample livox_lidar_quick_start by following instruction, and I got messages as below, and the actual lidar motor seems not running (I believe I put the right host IP and device IP in the config file).

```
[2022-10-12 21:24:26.542] [console] [info] Data Handler Init Succ.  [data_handler.cpp] [Init] [42]
[2022-10-12 21:24:26.542] [console] [info] Create detection channel detection socket:12  [device_manager.cpp] [CreateDetectionChannel] [223]
[2022-10-12 21:24:26.542] [console] [info] Create detection channel detection socket:12  [device_manager.cpp] [CreateDetectionChannel] [232]
``` 

The log says the detection creation was successful, but seems it's on IDLE mode not MOTOR_STARTUP mode according to the Livox SDK Communication Protocol HAP documentation. Is there a way to start-up the motor and get the pointcloud data? 

Thank you!

redheli, 2022-10-13:
show your config file and the command you use
and try ping the lidar ip?

changkk, 2022-10-13:
Hi, thanks for the reply! My config file looks as below:
```
{
	"HAP": {
		"device_type" : "HAP",
		"lidar_ipaddr": "192.168.1.100",
		"lidar_net_info" : {
			"cmd_data_port": 56000,
			"push_msg_port": 0,
			"point_data_port": 57000,
			"imu_data_port": 58000,
			"log_data_port": 59000
		},
		"host_net_info" : {
			"cmd_data_ip" : "192.168.12.37",
			"cmd_data_port": 56000,
			"push_msg_ip": "192.168.12.37",
			"push_msg_port": 0,
			"point_data_ip": "192.168.12.37",
			"point_data_port": 57000,
			"imu_data_ip" : "192.168.12.37",
			"imu_data_port": 58000,
			"log_data_ip" : "192.168.12.37",
			"log_data_port": 59000
		}
	}
}	
```

And also I can ping this ip:
```
PING 192.168.1.100 (192.168.1.100) 56(84) bytes of data.
64 bytes from 192.168.1.100: icmp_seq=1 ttl=64 time=0.037 ms
64 bytes from 192.168.1.100: icmp_seq=2 ttl=64 time=0.039 ms
```

redheli, 2022-10-13:
Seems not same subnet
Change your static ip to 192.168.1.xx

changkk, 2022-10-14:
Thank you! I tried to change the host static ip to 192.168.1.37, but no luck. Any other thoughts?

closed 2022-10-18 by changkk