Livox-SDK/Livox-SDK2 #10: LIVOX AVIA

fghoussen, 2022-12-22:
Does this SDK support AVIA LIVOX Lidar?
Can this SDK be used to get data (points) from AVIA LIVOX LIDAR? Is this only for HAP and Mid360?

Livox-Infra, 2023-01-13:
Sorry, Livox SDK2 support only HAP and Mid-360. For AVIA, pls refer to [Livox SDK](https://github.com/Livox-SDK/Livox-SDK)

fghoussen, 2023-01-13:
Does this means that AVIA is no more supported (as latest SDK - SDK2 - doesn't support it)?

Livox-Infra, 2023-01-16:
I'm afraid ... yes... Since Livox Driver2 depends on the latest Livox SDK2 and the latter doesn't support AVIA anymore.
Another solution is introducing Livox SDK1 into Livox ROS Driver2 but we do not have a plan to do that, yet...

seajayshore, 2023-01-25:
Is the Avia becoming end-of-life in general? 
Or will you continue to produce & support it with the older SDK still?

seajayshore, 2023-05-23:
It is confirmed that Avia is discontinued and we shouldn’t expect more stock:
https://forum.livoxtech.com/forum.php?mod=viewthread&tid=327&extra=page%3D1&page=2&mobile=2

You can also email Livox support directly and they will confirm.