Livox-SDK/Livox-SDK2 #14: how can i connect multiple livox mid 360 and hap to SDK 2

kiri666, 2023-01-25:
I want to use multiple Livox mid 360 and hap in one program. How can i do it? How do i have to setup the config file for that? How many mid 360 and hap can i use in one program?

Livox-Infra, 2023-02-15:
Just simply mix the config files of MID-360 & HAP, like this:
{
  HAP: {
         ...
  },
  MID360: {
         ...
   }
}
then MID-360s & HAPs can work together in one single app.

For the maximum supported Lidars number, it depens on the network condition and computer capabilities.

kiri666, 2023-02-16:
Thank you for the answer.
That helps me already. How is it now, if I have for example 10 mid360 sensors in the network but only want to scan with 5. In SDK 1 you could create a broadcast code list with the devices you wanted to use, then the SDK only connected to the defined sensors. 
Is it also possible in the SDK 2 to use only certain sensors and how?

zuomaidi52, 2023-02-22:
If i want to connect two Mid-360 in one Process. What should I do in config or cpp file.

Livox-Infra, 2023-03-13:
> Thank you for the answer. That helps me already. How is it now, if I have for example 10 mid360 sensors in the network but only want to scan with 5. In SDK 1 you could create a broadcast code list [...]

For this senario, you should modify the source code to filter by lidar ip.
We will come up with annother way to do such filter work via the config file, soon.