Livox-SDK/Livox-SDK2 #17: Building livox sdk2 in dockerr

jenamax, 2023-02-02:
Hello! I am trying to build LivoxSDK2 inside the docker container in the starting script by the command "cmake .. && make -j &&  make install" in build directory. And get the following error during the execution of "make install":

```
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/liblivox_lidar_sdk_static.a
CMake Error at sdk_core/cmake_install.cmake:41 (file):
  file INSTALL cannot copy file
  "/home/sfd/projects/self-driving-worker/self-driving-system/integration_3/src/livox/Livox-SDK2/build/sdk_core/liblivox_lidar_sdk_static.a"
  to "/usr/local/lib/liblivox_lidar_sdk_static.a": Permission denied.
Call Stack (most recent call first):
  cmake_install.cmake:42 (include)

```

How can I build the livox SDK in docker? I know that it is possible to run container with sudo privileges but this may be unsafe since it gives sudo permission to all components which are running in container. Is there some other option?

zuomaidi52, 2023-02-22:
Are you use `sudo`to enter docker? Also you could try `sudo make install`

Livox-Infra, 2023-03-13:
> How can I build the livox SDK in docker? I know that it is possible to run container with sudo privileges but this may be unsafe since it gives sudo permission to all components which are running in container. Is there some other option?

as @zuomaidi52 said, you just need to add `sudo` when installing the SDK2, `sudo make install`. No need to run the container with sudo privileges.