Livox-SDK/Livox-SDK2 #33: livox_lidar_quick_start crashes on Rasbpian Wheezy 32

michalpelka, 2023-04-23:
I experience random crashes due to UB on SDK example. The crash occurs every time I use livox_lidar_quick_start (it is little more stable when I use SDK). Here is stack for crashed `livox_lidar_quick_start`:
```
livox::lidar::GeneralCommandHandler::Init general_command_handler.cpp:86
livox::lidar::DeviceManager::Init device_manager.cpp:126
LivoxLidarSdkInit livox_lidar_sdk.cpp:95
main basic_string.h:186
__libc_start_main 0x00000000f7c89740
_start 0x00000000000397dc
```
Here is my configuration:
```
robot@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

michalpelka, 2023-05-06:
Any update on that issue?
It is seems quite serious issue.

neolu, 2023-05-19:
Raspberry Pi OS (buster) is ok.

michalpelka, 2023-05-19:
For me, it worked as well for some time. After updating the system and recompilation it stops working. It is UB triggered by padding of non-trivial classes. 
As I said it is a serious bug that causes SDK to be unusable for me as a Livox client. I do not understand why it was not addressed.

michalpelka, 2023-05-19:
@Livox-Infra @Livox-SDK

JanuszBedkowski, 2023-05-25:
I confirm the same issue on rasbian. This is serious blocker.

HViktorTsoi, 2023-08-31:
This commit https://github.com/Livox-SDK/Livox-SDK2/pull/34 fixed my bus error issue on **raspberry pi 10 (buster)**.

Thanks @michalpelka 

In short, just modify the following two lines of code

https://github.com/Livox-SDK/Livox-SDK2/blob/5c1f7660b6daae2420fc4b8132c1c3f3c12c0756/sdk_core/logger_handler/logger_manager.h#L44
and
https://github.com/Livox-SDK/Livox-SDK2/blob/5c1f7660b6daae2420fc4b8132c1c3f3c12c0756/sdk_core/comm/define.h#L40

both to

```c++
#pragma pack(4)
```

michalpelka, 2023-09-04:
@Livox-Infra @Livox-SDK could you approve or #34 or the solution proposed by @HViktorTsoi ?

RomanStadlhuber, 2024-06-22:
@michalpelka thank you for this fix. I am also experiencing the same problem, trying to get [livox_ros_driver2](https://github.com/Livox-SDK/livox_ros_driver2) to work
Simply trying to refactor the `#pragma pack(x)` statements did not do the trick. May I ask what is the reasoning behind this fix so I can maybe better understand what is going on in the ROS driver?

michalpelka, 2024-06-24:
Ok, @RomanStadlhuber I spent a quite long time on that, and I am more than unhappy that it was closed by maintainers, without fixing.

I was experiencing a crash only with Raspberry Pi with a 32 and 64-bit system. It worked as expected on x86.
The crash was quite strange, instead of getting seg-fault we got bus-error. And stack that was not usable.
The difference between seg-fault and bus-error is here : https://stackoverflow.com/questions/212466/what-is-a-bus-error-is-it-different-from-a-segmentation-fault

I guess that it is not a bug in Livox-SDK, but rather some bug in the compiler or an advanced example of misuage of alignment. ARM by design has to have instructions aligned to 4 bytes (https://stackoverflow.com/questions/1237963/alignment-along-4-byte-boundaries) - it will stop execution if we ask ARM to run instructions on not aligned address. I did not analyze the assembly, but I am almost sure that `pragma pack(1)`  was also applied globally to classes.
Note that using `pack(4)` will align everything to the accepted 4 bytes alignment for ARMs.

michalpelka, 2024-06-24:
Also what was very strange, it was compile-time undefined behavior - it worked fine (it did not crash randomly), but adding anything (e.g. another variable) in another place in the program that was using the SDK caused persistent crashes.

michalpelka, 2024-06-24:
Debugging this issue was quite an adventure and learning experience for me.

RomanStadlhuber, 2024-07-02:
@michalpelka Thank you for your answer. I fixed it and it was, in fact, just a single `#pragma pack` statement instead of both (see [Issue for reference](https://github.com/Livox-SDK/livox_ros_driver2/issues/130#issuecomment-2204033674)).
This fix would not be possible if it wasn't for the problem you discovered, many thanks!

michalpelka, 2024-07-02:
Glad my investigation helped me.