Livox-SDK/Livox-SDK2 #39: livox HAP(tx) 时间同步如何实现？

NHMMing, 2023-05-22:
您好，目前我想对已有的HAP(tx)来进行时间同步。查看手册后发现该雷达支持的是gPTP协议的时间同步方式，我们的系统有一个稳定的GPS时钟源，目前想把雷达的时间戳也同步至该GPS时钟上，允许的误差在1-2ms。想问下我该怎么样去实现这个方案？软件硬件上分别需要做些什么，希望得到您的回复。

NHMMing, 2023-05-23:
官方能给个回复吗

Livox-Infra, 2023-05-23:
Hi @NHMMing , 目前 HAP 的硬件版本缺少接收 GPS 时钟信号的模块，所以并无直接的方案让 HAP 与 GPS 时钟同步；

或许可尝试用 PC 做中转，让 HAP 与一台对齐了 GPS 时间的主机进行同步。
然而我们也没尝试过这种方式，不确保可行性，所以仅供参考

NHMMing, 2023-07-24:
忘记更新，您回复之后我尝试了您说的方法，可以正常拿到点云帧的绝对时间。时间精度取决于给主机的授时精度以及gptp同步精度。

closed 2023-10-24 by NHMMing

zymouse, 2024-04-15:
Hello, use[livox HUP](https://www.livoxtech.com/3296f540ecf5458a8829e01cf429798e/downloads/20190218/Livox%20Hub%20Quick%20Start%20Guide_v1.0_20190218.pdf)。
[image]

1.0 Is it possible to synchronize GPS time for `mid360` and `hap`?
2.0 Can `livox HUP` be triggered for synchronization of `camra`, `mid360`, `hap`?