Livox-SDK/Livox-SDK2 #42: hap

wbq13599531791, 2023-06-01:
I want to use HAP to receive data,and run the example "livox lidar quick start",but it has the problem "[console] [error] Query livox lidar Fw type failed, the status:-4  [general_command_handler.cpp] [QueryFwTypeCallback] [438]",can you tell me why result in this problem and how to handle it?

DerK96, 2023-08-11:
What is your firmware version?
I had the same issue with firmware version 15.10.01.08 and updated to the latest version 15.10.01.15, that solved the issue for me.

wuruixia1993, 2023-09-29:
I have encountered the same problem as you. Can you provide me with a specific solution? Thank you

DerK96, 2023-09-30:
did you update your firmware to the latest version?

stevencolinmartin, 2023-10-17:
Updated firmware also resolved this for me.

delipl, 2023-11-02:
How to upgrade firmware?

> Updated firmware also resolved this for me.

reddittoob, 2023-11-16:
I was upgrade firmware to the latest version. However, I got same error. 

What I have to do?

SVMawww, 2023-11-22:
hello,I incountered this problem too,did you solve it?

wuruixia1993, 2023-11-23:
不好意思，刚看到你的邮件,我的问题已解决
ros主从机设置时网关为192.16.0.1，激光雷达的ip设置时网关也被设置成了192.168.0.1
切换网络后，ros主从机的网关更改，报错解决。
问题应该是ros主从机的网关与激光雷达冲突，更改其中一个即可

SVMawww, 2023-11-23:
感谢回复，我也解决了问题

reddittoob, 2023-11-24:
How solve this problem?

reddittoob, 2023-11-24:
what does it mean switching the network?

I don't understand your solution. Please let me know about details.

reddittoob, 2023-11-24:
I don't understand you said. 

Can you let me know about details?

DDQXZcp, 2023-12-14:
I encountered this problem and I solved it in another way. I set my host IP to 192.168.1.50 but it doesn't work. Then I found that the manual said you also need to set the subnet mask to "255.255.255.0". Then it works.

reddittoob, 2023-12-14:
I already set subnet mask to "255.2255.255.0". But I still have same problem.

wuruixia1993, 2023-12-27:
> I already set subnet mask to "255.255.255.0". But I still have same problem.

不好意思，刚看到你的邮件,我的问题已解决
ros主从机设置时网关为192.16.0.1，激光雷达的ip设置时网关也被设置成了192.168.0.1
切换网络后，ros主从机的网关更改，报错解决。
问题应该是ros主从机的网关与激光雷达冲突，更改其中一个即可

XuechenCao, 2024-11-14:
i also meet the problem , iu set the correct the IP, but the SDK still report error:
[2024-11-14 05:34:54.046] [console] [error] Query livox lidar Fw type failed, the status:-4  [general_command_handler.cpp] [QueryFwTypeCallback] [407]

wuruixia1993, 2024-11-29:
网关地址冲突，你把连接wifi的地址不要设置为192.168.1.   和激光雷达的设置不一样即可