Livox-SDK/Livox-SDK2 #44: Segfault at exit in spdlog destructors

and502, 2023-06-24:
Hi,

I'm using this library in a ROS2 node and getting a segfault during destruction of spdlog classes at shutdown. Please see valgrind output below. This seems to be caused by the ros2 node binary dynamically linking to and using my system spdlog librares, while the Livox SDK is compiled using the emmbedded spdlog headers in the src/3rdparty/spdlog/ directory. I've fixed this by disabling the use of  src/3rdparty/spdlog/ and compiling the SDK using headers from my system install of spdlog.

`
lds destory!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
==240260== Invalid read of size 8
==240260==    at 0x514F9FA: std::_Sp_counted_ptr_inplace<spdlog::logger, std::allocator<spdlog::logger>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (in /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2)
==240260==    by 0x10FB0E: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (in /home/and502/lcs/lc-slam/livox/livox/build/driver/install/livox_ros_driver2/lib/livox_ros_driver2/livox_ros_driver2_node)
==240260==    by 0x6D03946: std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() (in /home/and502/lcs/lc-slam/livox/livox/build/driver/install/livox_ros_driver2/lib/liblivox_ros_driver2.so)
==240260==    by 0x6F9E431: std::__shared_ptr<spdlog::logger, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() (in /home/and502/lcs/lc-slam/livox/livox/build/sdk/install/lib/liblivox_lidar_sdk_shared.so)
==240260==    by 0x6F9E451: std::shared_ptr<spdlog::logger>::~shared_ptr() (in /home/and502/lcs/lc-slam/livox/livox/build/sdk/install/lib/liblivox_lidar_sdk_shared.so)
==240260==    by 0x4D41494: __run_exit_handlers (exit.c:113)
==240260==    by 0x4D4160F: exit (exit.c:143)
==240260==    by 0x4D25D96: (below main) (libc_start_call_main.h:74)
==240260==  Address 0xb4294e0 is 0 bytes inside a block of size 64 free'd
==240260==    at 0x484B8AF: operator delete(void*) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==240260==    by 0x5154E29: std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<spdlog::logger> >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<spdlog::logger> > >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::clear() (in /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2)
==240260==    by 0x514A251: spdlog::details::registry::~registry() (in /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2)
==240260==    by 0x4D41494: __run_exit_handlers (exit.c:113)
==240260==    by 0x4D4160F: exit (exit.c:143)
==240260==    by 0x4D25D96: (below main) (libc_start_call_main.h:74)
==240260==  Block was alloc'd at
==240260==    at 0x4849013: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==240260==    by 0x51584E5: std::__detail::_Map_base<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<spdlog::logger> >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<spdlog::logger> > >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true>, true>::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2)
==240260==    by 0x514C8B0: spdlog::details::registry::register_logger_(std::shared_ptr<spdlog::logger>) (in /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2)
==240260==    by 0x514C9D8: spdlog::details::registry::register_logger(std::shared_ptr<spdlog::logger>) (in /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2)
==240260==    by 0x514CAEB: spdlog::register_logger(std::shared_ptr<spdlog::logger>) (in /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2)
==240260==    by 0x6FED364: InitLogger() (in /home/and502/lcs/lc-slam/livox/livox/build/sdk/install/lib/liblivox_lidar_sdk_shared.so)
==240260==    by 0x6FCF90B: LivoxLidarSdkInit (in /home/and502/lcs/lc-slam/livox/livox/build/sdk/install/lib/liblivox_lidar_sdk_shared.so)
==240260==    by 0x6D560BF: livox_ros::LdsLidar::InitLivoxLidar() (in /home/and502/lcs/lc-slam/livox/livox/build/driver/install/livox_ros_driver2/lib/liblivox_ros_driver2.so)
==240260==    by 0x6D55EF6: livox_ros::LdsLidar::InitLidars() (in /home/and502/lcs/lc-slam/livox/livox/build/driver/install/livox_ros_driver2/lib/liblivox_ros_driver2.so)
==240260==    by 0x6D55E1E: livox_ros::LdsLidar::InitLdsLidar(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/and502/lcs/lc-slam/livox/livox/build/driver/install/livox_ros_driver2/lib/liblivox_ros_driver2.so)
==240260==    by 0x6CFFF87: livox_ros::DriverNode::DriverNode(rclcpp::NodeOptions const&) (in /home/and502/lcs/lc-slam/livox/livox/build/driver/install/livox_ros_driver2/lib/liblivox_ros_driver2.so)
==240260==    by 0x6D0DC6A: void __gnu_cxx::new_allocator<livox_ros::DriverNode>::construct<livox_ros::DriverNode, rclcpp::NodeOptions const&>(livox_ros::DriverNode*, rclcpp::NodeOptions const&) (in /home/and502/lcs/lc-slam/livox/livox/build/driver/install/livox_ros_driver2/lib/liblivox_ros_driver2.so)
==240260==
==240260== Invalid read of size 8
==240260==    at 0x514FA01: std::_Sp_counted_ptr_inplace<spdlog::logger, std::allocator<spdlog::logger>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (in /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2)
==240260==    by 0x10FB0E: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (in /home/and502/lcs/lc-slam/livox/livox/build/driver/install/livox_ros_driver2/lib/livox_ros_driver2/livox_ros_driver2_node)
==240260==    by 0x6D03946: std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() (in /home/and502/lcs/lc-slam/livox/livox/build/driver/install/livox_ros_driver2/lib/liblivox_ros_driver2.so)
==240260==    by 0x6F9E431: std::__shared_ptr<spdlog::logger, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() (in /home/and502/lcs/lc-slam/livox/livox/build/sdk/install/lib/liblivox_lidar_sdk_shared.so)
==240260==    by 0x6F9E451: std::shared_ptr<spdlog::logger>::~shared_ptr() (in /home/and502/lcs/lc-slam/livox/livox/build/sdk/install/lib/liblivox_lidar_sdk_shared.so)
==240260==    by 0x4D41494: __run_exit_handlers (exit.c:113)
==240260==    by 0x4D4160F: exit (exit.c:143)
==240260==    by 0x4D25D96: (below main) (libc_start_call_main.h:74)
==240260==  Address 0xb4294d8 is 8 bytes before a block of size 64 free'd
==240260==    at 0x484B8AF: operator delete(void*) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==240260==    by 0x5154E29: std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<spdlog::logger> >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::shared_ptr<spdlog::logger> > >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::clear() (in /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2)
==240260==    by 0x514A251: spdlog::details::registry::~registry() (in /usr/lib/x86_64-linux-gnu/libspdlog.so.1.9.2)
==240260==    by 0x4D41494: __run_exit_handlers (exit.c:113)
==240260==    by 0x4D4160F: exit (exit.c:143)
[... cut ...]
==240260==    by 0x4D41494: __run_exit_handlers (exit.c:113)
==240260==    by 0x4D4160F: exit (exit.c:143)
==240260==    by 0x4D25D96: (below main) (libc_start_call_main.h:74)
==240260==  If you believe this happened as a result of a stack
==240260==  overflow in your program's main thread (unlikely but
==240260==  possible), you can try to increase the size of the
==240260==  main thread stack using the --main-stacksize= flag.
==240260==  The main thread stack size used in this run was 8388608.
`

NHMMing, 2023-07-24:
I have problem similar to this. just use "thirdparty/spdlog" the SDK provided

NHMMing, 2023-08-09:
> > I have problem similar to this. just use "thirdparty/spdlog" the SDK provided
> 
> Will appreciate if could give more detail operations

thanks, I will upload more log info about this problem.

landersson, 2023-08-24:
I'm using other libraries that are also linking to the system version of spdlog, which means that I can't fix this by making these libraries link to the spdlog version embedded in the livox_sdk2 library.

One possible fix for this would be to add a cmake option to enable compiling the sdk against the global, system wide installation of spdlog instead of the one included in the Livox-SDK2 repo.