Livox-SDK/Livox-SDK2 #46: How to use the SDK2(our tool is HAP Livox Lidar) in LIO-Livox ???

17863958533, 2023-07-23:
We want to test the HAP Lidar in LIO-Livox, and we think the SDK2 environment is OK. Our environment is Ubuntu18.04 and ROS1 Meliodic.
When we  run the SDK2 in this "livox_lidar_quick_start", our HAP Lidar has been working.
Then, we run the "roslaunch livox_ros_driver livox_lidar_msg.launch" and "roslaunch lio_livox hap.launch", the code run OK.
After that, we can see the rviz window, but ,but! The Rviz window cannot see pointcloud data! 
We want to know more details steps about using HAP in LIO-Livox.
If you know this problem, you can also send e-mail: 2310267@tongji.edu.cn

muaTomato, 2023-08-24:
Have you solved it please, I have the same problem!

Shidabot, 2024-07-17:
> We want to test the HAP Lidar in LIO-Livox, and we think the SDK2 environment is OK. Our environment is Ubuntu18.04 and ROS1 Meliodic. When we run the SDK2 in this "livox_lidar_quick_start", our HAP Lidar has been working. Then, we run the "roslaunch livox_ros_driver livox_lidar_msg.launch" and "roslaunch lio_livox hap.launch", the code run OK. After that, we can see the rviz window, but ,but! The Rviz window cannot see pointcloud data! We want to know more details steps about using HAP in LIO-Livox. If you know this problem, you can also send e-mail: [2310267@tongji.edu.cn](mailto:2310267@tongji.edu.cn)

厉博，livox_lidar_msg.launch发布Customsg的格式，rviz需要pointcloud2的格式