Livox-SDK/Livox-SDK2 #49: 当连接多个雷达时，如何区分数据包来自于哪个雷达？

NHMMing, 2023-08-03:
@Livox-Infra @Livox-SDK 
目前我连接了四台HAP，同时扫描和存储数据。我想区分四个雷达的数据，然后将对应雷达的数据存储到目标文件夹。目前我是通过在接收packer时，通过句柄来判断当前packet是来自不同的雷达并分别进行存储。但现在的问题是我无法通过此方法将接收到的来自四个不同雷达的数据准确对应到是我的雷达1、雷达2、雷达3、雷达4，因为这个过程中存在随机性，无法像sn号那样将句柄和雷达硬件提前对应起来。请问官方有什么好的方法可以帮助我解决这个问题吗？期待您的回复

NHMMing, 2023-08-03:
@Livox-Infra 官方可以给个回复吗

NHMMing, 2023-08-04:
@Livox-Infra @Livox-SDK 官方看见了吗？想得到您的答复

NHMMing, 2023-08-07:
@Livox-Infra 官方可以回复下吗？

pjnhlh, 2023-08-07:
> 

hi @NHMMing , 可以按照 handle 区分雷达，实际上 handle 是雷达 IP 的整数表达形式。劳烦暂先参考 https://github.com/Livox-SDK/livox_ros_driver2/blob/master/src/comm/comm.cpp 中的 std::string IpNumToString(uint32_t ip_num) 的实现代码将 handle 转换为 IP 字符串。

我们尽快在 SDK2 里提供直接的转换 API

pjnhlh, 2023-08-07:
hi @NHMMing , 可以按照 handle 区分雷达，实际上 handle 是雷达 IP 的整数表达形式。可以参考 https://github.com/Livox-SDK/livox_ros_driver2/blob/master/src/comm/comm.cpp 中的 std::string IpNumToString(uint32_t ip_num) 的实现代码将 handle 转换为 IP 字符串。

NHMMing, 2023-08-07:
> hi @NHMMing , 可以按照 handle 区分雷达，实际上 handle 是雷达 IP 的整数表达形式。可以参考 https://github.com/Livox-SDK/livox_ros_driver2/blob/master/src/comm/comm.cpp 中的 std::string IpNumToString(uint32_t ip_num) 的实现代码将 handle 转换为 IP 字符串。

十分感谢回复！我会尝试使用上述方法

closed 2023-08-07 by NHMMing

samsaber1998, 2024-03-25:
楼主你好，请问你是如何获得每一帧的点云数据包