Livox-SDK/Livox-SDK2 #55: MID 360 LiDAR Integration in ROS2

Prabuddhi-05, 2023-10-12:
I am currently working on integrating the MID 360 LiDAR into a ROS 2 environment. I have successfully operated the LiDAR with Windows application and ROS1.

Here's a brief overview of my setup and the problem I'm facing:

Setup Details:
Host IP: 192.168.1.50
LiDAR IP: 192.168.1.102

[image]

ROS 2 Environment:
OS: Ubuntu 22.04
ROS 2 Distribution: Humble

Software Installation:
Installed Livox SDK 2 from https://github.com/Livox-SDK/Livox-SDK2
Installed livox_ros_driver2 from https://github.com/Livox-SDK/livox_ros_driver2

Configuration:
Edited the IP addresses for both the host and LiDAR in the MID360_config.json file to match the IP settings.

[image]

Problem Description:
When I launch the rviz_MID360_launch.py script, which is intended to configure RViz for visualizing LiDAR data, I encounter an issue: the point cloud is not displayed in RViz.

[image]

I've performed various checks, such as verifying network connectivity, and ensuring the accuracy of the IP configuration. However, I haven't been able to resolve the issue on my own.

closed 2023-10-19 by Prabuddhi-05

Prabuddhi-05, 2023-10-23:
The issue was resolved by turning off the firewall to ensure that some of the ports were no longer blocked.

SurendhranV, 2025-06-11:
HI ,
i used livox mid 360 in ros2 humble but getting some error like bind error if anybody know please tell me:
user@user:~/ws_livox/src$ ros2 launch livox_ros_driver2 rviz_MID360_launch.py
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2025-06-10-17-51-59-952149-user-19932
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [livox_ros_driver2_node-1]: process started with pid [19933]
[INFO] [rviz2-2]: process started with pid [19935]
[livox_ros_driver2_node-1] [INFO] [1749558120.043799958] [livox_lidar_publisher]: Livox Ros Driver2 Version: 1.2.4
[livox_ros_driver2_node-1] [INFO] [1749558120.044274461] [livox_lidar_publisher]: Data Source is raw lidar.
[livox_ros_driver2_node-1] [INFO] [1749558120.044300504] [livox_lidar_publisher]: Config file : /home/user/ws_livox/install/livox_ros_driver2/share/livox_ros_driver2/launch_ROS2/../config/MID360_config.json
[livox_ros_driver2_node-1] LdsLidar *GetInstance
[livox_ros_driver2_node-1] config lidar type: 8
[livox_ros_driver2_node-1] successfully parse base config, counts: 1
[livox_ros_driver2_node-1] bind failed
[livox_ros_driver2_node-1] Failed to init livox lidar sdk.
[livox_ros_driver2_node-1] [ERROR] [1749558120.047151608] [livox_lidar_publisher]: Init lds lidar fail!