Livox-SDK/Livox-SDK2 #60: SetLivoxLidarPointCloudCallBack does not sample. How can I obtain all point cloud data for each frame?

Waynepoo, 2023-11-20:
I am currently working with the Livox Lidar Quick Start project, and I have a question regarding the SetLivoxLidarPointCloudCallBack function.

In my setup, I am using the callback function PointCloudCallback, and I have observed that data->dot_num only provides 96 sample points. However, I am interested in obtaining the complete set of points for a single frame

georgflick, 2023-11-21:
I think for that you can have a look how its done in the [livox_ros_driver2](https://github.com/Livox-SDK/livox_ros_driver2), there the points are collected in a queue (in the OnLivoxLidarPointCloudCallback function) and a timer (running with e.g. 10 Hz) is then taking all the collected points and publishes them on a rostopic.

samsaber1998, 2024-03-25:
Hi, do u find the way to get pointcloud data for each frame? I have the same problem

lmz901, 2024-08-01:
Same problem. Does anyone have a solution?

AlphaNugie, 2025-02-15:
The same problem has haunted me for days, but as I read the manual I found out that there's no such thing as "each frame" or "single frame" for HAP lidar, because according to the manual, the longer you waited while collecting the sample points in each package, the more details will be revealed. Kind of like the concept of exposure in photography.