Livox-SDK/Livox-SDK2 #62: Failed to init livox lidar sdk. bind failed

XinCheng-lee, 2023-11-29:
运行：roslaunch livox_ros_driver2 rviz_MID360.launch

错误：

bind failed
[2023-11-29 23:05:22.214] [console] [error] Create detection socket failed.  [device_manager.cpp] [CreateDetectionChannel] [275]
[2023-11-29 23:05:22.214] [console] [error] Create detection channel failed.  [device_manager.cpp] [CreateChannel] [242]
[2023-11-29 23:05:22.214] [console] [error] Create channel failed.  [device_manager.cpp] [Init] [169]
Failed to init livox lidar sdk.
[ERROR] [1701270322.214576229]: Init lds lidar failed!

magic-Alex, 2023-11-30:
How did you fix this？你是如何解决的，可以告知一下吗？

XinCheng-lee, 2023-11-30:
> How did you fix this？你是如何解决的，可以告知一下吗？

livox_ros_driver2要装对应的Livox-SDK2而不是Livox-SDK。如果你用的是livox_ros_driver，就装Livox-SDK。

magic-Alex, 2023-11-30:
> > How did you fix this？你是如何解决的，可以告知一下吗？
> 
> livox_ros_driver2要装对应的Livox-SDK2而不是Livox-SDK。如果你用的是livox_ros_driver，就装Livox-SDK。

您是livox_ros_driver2嘛，我按步骤安装了livox_ros_driver2和Livox-SDK2可是运行仍然提示
[livox_ros_driver2_node-1] LdsLidar *GetInstance
[livox_ros_driver2_node-1] config lidar type: 8
[livox_ros_driver2_node-1] successfully parse base config, counts: 1
[livox_ros_driver2_node-1] bind failed
[livox_ros_driver2_node-1] Failed to init livox lidar sdk.
[livox_ros_driver2_node-1] [ERROR] [1701325485.539084397] [livox_lidar_publisher]: Init lds lidar fail!

closed 2023-12-01 by XinCheng-lee

magic-Alex, 2023-12-01:
> > > > > > How did you fix this？你是如何解决的，可以告知一下吗？
> > > > > 
> > > > > 
> > > > > livox_ros_driver2要装对应的Livox-SDK2而不是Livox-SDK。如果你用的是livox_ros_driver，就装Livox-SDK。
> > > > 
> > > > 
> > > > 您是livox_ros_driver2嘛，我按步骤安装了livox_ros_driver2和Livox-SDK2可是运行仍然提示 [livox_ros_driver2_node-1] LdsLidar *GetInstance [livox_ros_driver2_node-1] config lidar type: 8 [livox_ros_driver2_node-1] successfully parse base config, counts: 1 [livox_ros_driver2_node-1] bind failed [livox_ros_driver2_node-1] Failed to init livox lidar sdk. [livox_ros_driver2_node-1] [ERROR] [1701325485.539084397] [livox_lidar_publisher]: Init lds lidar fail!
> > > 
> > > 
> > > livox_ros_driver2里的config文件要改一下ip
> > 
> > 
> > { "lidar_summary_info" : { "lidar_type": 8 }, "HAP": { "lidar_net_info" : { "cmd_data_port": 56000, "push_msg_port": 0, "point_data_port": 57000, "imu_data_port": 58000, "log_data_port": 59000 }, "host_net_info" : { "cmd_data_ip" : "192.168.8.117", "cmd_data_port": 56000, "push_msg_ip": "", "push_msg_port": 0, "point_data_ip": "192.168.8.117", "point_data_port": 57000, "imu_data_ip" : "192.168.8.117", "imu_data_port": 58000, "log_data_ip" : "", "log_data_port": 59000 } }, "lidar_configs" : [ { "ip" : "192.168.8.100", "pcl_data_type" : 1, "pattern_mode" : 0, "extrinsic_parameter" : { "roll": 0.0, "pitch": 0.0, "yaw": 0.0, "x": 0, "y": 0, "z": 0 } } ] }
> > 电脑的ip是192.168.8.117，我不知道网段是否有影响，但是我使用LivoxViewer2工具是可以搜索到雷达的，您有什么建议嘛？
> 
> 你看一下readme里面的LiDAR config，另外你用的是mid360还是HAP？

我是HAP和mid360混合使用，已得到解决，确实是IP配置问题，感谢您

proxi666, 2024-02-20:
@magic-Alex 

SDK2文件夹或livox_ros_driver2文件夹中的IP配置问题？

faithg3, 2024-06-25:
> > > > > > > How did you fix this？你是如何解决的，可以告知一下吗？
> > > > > > 
> > > > > > 
> > > > > > livox_ros_driver2 install the corresponding Livox-SDK2 instead of Livox-SDK. If you're using livox_ros_driver, install Livox-SDK.
> > > > > 
> > > > > 
> > > > > 您是livox_ros_driver2嘛，我按步骤安装了livox_ros_driver2和Livox-SDK2可是运行仍然提示 [livox_ros_driver2_node-1] LdsLidar *GetInstance [livox_ros_driver2_node-1] config lidar type: 8 [livox_ros_driver2_node-1] successfully parse base config, counts: 1 [livox_ros_driver2_node-1] bind failed [livox_ros_driver2_node-1] Failed to init livox lidar sdk. [livox_ros_driver2_node-1] [ERROR] [1701325485.539084397] [livox_lidar_publisher]: Init lds lidar fail!
> > > > 
> > > > 
> > > > The config file in the livox_ros_driver2 needs to be changed to the IP
> > > 
> > > 
> > > { "lidar_summary_info" : { "lidar_type": 8 }, "HAP": { "lidar_net_info" : { "cmd_data_port": 56000, "push_msg_port": 0, "point_data_port": 57000, "imu_data_port": 58000, "log_data_port": 59000 }, "host_net_info" : { "cmd_data_ip" : "192.168.8.117", "cmd_data_port": 56000, "push_msg_ip": "", "push_msg_port": 0, "point_data_ip": "192.168.8.117", "point_data_port": 57000, "imu_data_ip" : "192.168.8.117", "imu_data_port": 58000, "log_data_ip" : "", "log_data_port": 59000 } }, "lidar_configs" : [ { "ip" : "192.168.8.100", "pcl_data_type" : 1, "pattern_mode" : 0, "extrinsic_parameter" : { "roll": 0.0, "pitch": 0.0, "yaw": 0.0, "x": 0, "y": 0, "z": 0 } } ] } The IP of the computer is 192.168.8.117, I don't know if the network segment has an impact, but I can search for radar with LivoxViewer2 tool, do you have any suggestions?
> > 
> > 
> > If you look at the LiDAR config in the readme, do you use mid360 or HAP?
> 
> I'm a mix of HAP and mid360 and it's been resolved, it's indeed an IP configuration issue, thank you

How did you fix the issue?

mpheng666, 2024-10-01:
Remember to change the config in the install path, it should look something like ws_livox/install/livox_ros_driver2/share/livox_ros_driver2/config/MID360_config.json