Livox-SDK/Livox-SDK2 #65: PPS signal support in Livox SDK2.

rahulsharma11, 2023-12-21:
Dear Team.

I am trying to use MID360 and camera synchronously. For that i am generating a 10Hz signal from external signal generator (arduino), and connecting to PPS pin of MID aviation wire.

Camera is getting synced but not MID360. (i am checking in LivoxViewer2).

Please clarify following:
1) Is PPS sync not supported?

2) For GPS syncing following steps are required?
     -> connect PPS signal to PPS pin.
     -> connect GPS output with the system where Lidar is connected.

Thanks.

michalpelka, 2024-01-06:
I was able to synchronize MiD-360 with https://github.com/michalpelka/esp2822_NMEA_sim.

YikangZhang1641, 2024-05-17:
> I was able to synchronize MiD-360 with https://github.com/michalpelka/esp2822_NMEA_sim.

Hello, I tried your method and successfully synchronized Mid360 with external STM32. 
However, as long as the sync cable is connected and state become "GPS Sync" in LivoxViewer2, the point cloud would periodically get stuck, 40sec in every 100sec to be specific. With cable removed, everything's fine.
Would this also happend to you?