Livox-SDK/Livox-SDK2 #69: STRANGE: cannot receive sensor data

Biruberuvinto, 2024-02-07:
So far, I have tried Livox Viewer, Livox-SDK2, and livox_ros2_driver, and confirmed that they all behave the same as the reference.

However, when I started up MID-360 for the first time in several months, I found that I was unable to receive sensor data, even though IP connectivity was confirmed.
This phenomenon was confirmed in Viewer, SDK Quick Start, and RViz.

I would like to perform an operation such as checking the health status of the MID-360 equipment in order to isolate the problem.Does anyone know?

Information is provided below for your reference.

**MID-360 information**
Firmware Version: 13.17.01.07
Hardware Version: 00.00.00.00
Loader Version: 13.14.99.9
Time Sync Type: No Sync

A warning is generated on the viewer with additional information.
> PPS time synchronization fail becuase of loss GPS signal
> Please check GPS signal
>
> GPS signal is abnormal
> Please check GPS time source

run livox_lidar_quick_start:
> ~/Livox-SDK2/build/samples/livox_lidar_quick_start$ ./livox_lidar_quick_start ../../../samples/livox_lidar_quick_start/mid360_config.json 
[2024-02-07 14:13:20.969] [console] [info] Livox lidar logger disable.  [parse_cfg_file.cpp] [Parse] [84]
[2024-02-07 14:13:20.969] [console] [info] Data Handler Init Succ.  [data_handler.cpp] [Init] [49]
[2024-02-07 14:13:20.969] [console] [info] Init livox lidars succ.  [device_manager.cpp] [Init] [153]

Biruberuvinto, 2024-02-07:
I have quoted, but the output should normally be as follows.
> [2023-01-20 12:02:04.523] [console] [info] Handle detection data, handle:1711384768, dev_type:9, sn:47MDKBV0010102, cmd_port:56100  [general_command_handler.cpp] [HandleDetectionData] [345]
point cloud handle: 1711384768, data_num: 96, data_type: 1, length: 1380, frame_counter: 0
point cloud handle: 1711384768, data_num: 96, data_type: 1, length: 1380, frame_counter: 0
point cloud handle: 1711384768, data_num: 96, data_type: 1, length: 1380, frame_counter: 0
point cloud handle: 1711384768, data_num: 96, data_type: 1, length: 1380, frame_counter: 0
[2023-01-20 12:02:04.525] [console] [info]  Receive Ack: Id 257 Seq 3  [mid360_command_handler.cpp] [Handle] [73]
[2023-01-20 12:02:04.525] [console] [info] Query Fw type succ, the fw_type:1  [general_command_handler.cpp] [QueryFwTypeCallback] [489]
point cloud handle: 1711384768, data_num: 96, data_type: 1, length: 1380, frame_counter: 0

arianizadi, 2024-02-16:
Did you end up finding the issue? I am running into the same problem.

arianizadi, 2024-02-16:
The issue ended up being the HOST PC and Lidar were not on the same subnet, for anyone looking at this in the future.

coolbotic, 2024-06-04:
Did you ever find a solution to this? I am having the exact same problem, cannot receive data on SDK, ROS or Viewer and getting the same logs as you for the SDK. I had it working on this host computer before, so not sure what has changed, still works on a different host computer so I'm guessing something to do with Ubuntu install but not sure?

YueqiUNSW, 2024-09-15:
Hey guys, I have encountered similar problems: my mid360 can't send data even if the IP configs are correct.
I turn off all the firewalls of Windows and it works. Hope this could help you guys.

AravindPrabhs, 2025-05-27:
I am seeing the exact same thing and can see point cloud packets being sent to the correct host port on Wireshark. I think the SDK is rejecting the packets but unsure why ? Anyone made any progress on this ?

arianizadi, 2025-05-27:
Just to clarify more specifically, what I did to solve my issue. When checking `ip a`, I see an ipv4 address for my wireless adapter, but my Ethernet (where the lidar is connected) did not have an IP address assigned. 

[image]

[image]

Let's say when opening the LivoxViewer I see the IP assigned to the mid-360 is `192.168.1.50/24`. 

If you are not sure what the /XX should be for yours, you can use a CIDR calculator or ChatGPT. 

I use `sudo ip addr add 192.168.1.50/24 dev eth0`, the command will assign this IP and subnet to my Ethernet adapter so the Linux box can have a way to communicate with the lidar (the same gateway and subnet is the key here).

I am not sure if it will fix everyone's issues, but hopefully someone.