Livox-SDK/Livox-SDK2 #70: [MID360] GPS time sync 0x04050002 error

pepisg, 2024-02-09:
Hi! 

I've been using the time synchronization feature in the Livox MID360 and I've seen that ocasionally I get a `0x04050002` error in LivoxViewer2.

The error description does not hint properly what may be wrong. Synchronization seems to work well regardless (as you can see from the timestamp in the image below), however I was still wondering what this error may be and how may I solve it.

Thanks in advance for your help!

[image]

YikangZhang1641, 2024-05-16:
Same to me! Have you solved it?

pepisg, 2024-05-16:
It looked like the error happened if I connected the GPS to the Lidar without having a proper fix, and then it would stay active in the viewer forever even after the fix was acquired