Livox-SDK/Livox-SDK2 #76: Ubuntu 24.04 Compatibility

JoLichtenfeld, 2024-05-08:
Hello,

is this SDK compatible with Ubuntu 24.04 / do you have plans to port it?

Thanks!

porizou, 2024-06-22:
Hi

I am using the LIVOX MID360 with Ubuntu 24.04 LTS and ROS2 Jazzy.
By adding <cstdint> to define.h and file_manager.h, I was able to build and use the Livox-SDK2 on Ubuntu 24.04.

```bash
diff --git a/sdk_core/comm/define.h b/sdk_core/comm/define.h
index 66996bf..fa4ce9f 100644
--- a/sdk_core/comm/define.h
+++ b/sdk_core/comm/define.h
@@ -33,6 +33,8 @@
 #include <atomic>
 
 #include "livox_lidar_def.h"
+#include <cstdint>  // add
+
 
 namespace livox {
 namespace lidar {
diff --git a/sdk_core/logger_handler/file_manager.h b/sdk_core/logger_handler/file_manager.h
index 242a431..21ac536 100644
--- a/sdk_core/logger_handler/file_manager.h
+++ b/sdk_core/logger_handler/file_manager.h
@@ -28,6 +28,7 @@
 #include <string>
 #include <vector>
 #include <map>
+#include <cstdint>  // add
```

JoLichtenfeld, 2024-06-23:
Thanks, works for me as well.

celaraze, 2024-07-01:
Very helpful, Thanks.

CxY-Nu, 2025-02-14:
@porizou Hi , this method also works for me, but when I was trying to build the ROS2 environment ,it failed .

./build.sh ROS2.  then shows CMake Error at CMakeLists.txt:224 (find Package). Do you have any ideas? thks!