Livox-SDK/Livox-SDK2 #77: Livox HAP: can't display pointcloud from Livox Viewer2 and ros driver

JunseoMin, 2024-05-16:
hi,
i connected livox HAP and my PC

PC: 192.168.1.2
HAP: 192.168.1.153

i think connection is stable but have some problem with 56000 udp port.

[image]

like screenshot, sending is available but can't receive data from lidar.

and the console logs are:
```
[2024.05.16-11.39.37:298][847]Livox: ULivoxUniformManager::QueryLidarInfo, time_sync_type changed [160] -> [0]
[2024.05.16-11.39.37:298][847]Livox: ULivoxUniformManager::QueryLidarInfo, FW_TYPE changed [106] -> [105]
[2024.05.16-11.39.37:298][847]Livox: ULivoxUniformManager::QueryLidarInfo, roi_cfg0 changed [1893392893 -2001731584 1893392893 1788870656] -> [1893392893 -2001731584 1893392893 1761607680]
[2024-05-16 20:39:37.358] [console] [info]  Receive Command: Id 258 Seq 155  [hap_command_handler.cpp] [Handle] [73]
[2024-05-16 20:39:37.381] [console] [info] Handle detection data, handle:2567022784, dev_type:10, sn:5CWD23984105SB1, cmd_port:56000  [general_command_handler.cpp] [HandleDetectionData] [348]
[2024-05-16 20:39:37.381] [console] [info] Host ip:192.168.1.2, lidar ip:192.168.1.153  [general_command_handler.cpp] [VerifyNetSegment] [292]
[2024-05-16 20:39:38.359] [console] [info]  Receive Command: Id 258 Seq 156  [hap_command_handler.cpp] [Handle] [73]
[2024-05-16 20:39:38.382] [console] [info] Handle detection data, handle:2567022784, dev_type:10, sn:5CWD23984105SB1, cmd_port:56000  [general_command_handler.cpp] [HandleDetectionData] [348]
[2024-05-16 20:39:38.382] [console] [info] Host ip:192.168.1.2, lidar ip:192.168.1.153  [general_command_handler.cpp] [VerifyNetSegment] [292]
[2024-05-16 20:39:39.292] [console] [info]  Receive Ack: Id 257 Seq 481  [hap_command_handler.cpp] [Handle] [70]
[2024-05-16 20:39:39.292] [console] [info]  Receive Ack: Id 257 Seq 482  [hap_command_handler.cpp] [Handle] [70]
```

i also can get data of lidar in livox viewer 2

[image]

but the pointcloud doesn't display,,,

plz let me know how can i solve this problem

thanks.

JunseoMin, 2024-05-17:
sloved

closed 2024-05-17 by JunseoMin

ofarukdemirtas, 2024-10-21:
how did u solve this issue? I have the same problem

JunseoMin, 2024-10-29:
In my case, the reason was a input voltage and input current. I solved it by converting input current