Livox-SDK/Livox-SDK2 #78: [mid360+pps+gprmc] repeating missing pointcloud packages for 40.0 sec in a 100.0s period

YikangZhang1641, 2024-05-16:
This happends frequently:

[image]

Before time 45.485, everything's OK. 10Hz Pointcloud and 200Hz IMU are well synchronized.
Something happens. 
After time 45.491, no Pointcloud but only IMU packages are transferred.

Pointcloud packages are resumed at time 85.484
[image]

With no hardware changed, the Lidar would be muted for exact 40s and awake for 60s, and then repeats in the next 100s loop: 

At time 145.495, pointclouds start to be missing again
[image]

until 185.488
[image]

miss at 245.495
[image]

resume at 285.493
[image]

However, with the PPS/GPRMC cable get disconnected and reconnected, it would not be an exact 100s loop.

It seems a firmware bug. I tested it on both windows and linux LivoxViewer, and on my code based on livox_ros_driver2.
Hope this would help.

closed 2024-05-17 by YikangZhang1641

YikangZhang1641, 2024-05-17:
Anyone who came from https://github.com/sheng00125/LIV_handhold or https://gitee.com/gwmunan/fast-livo_-reproduction trying to set hardware trigger might face the same problem.
The original GPRMC generating code in Timer.c might work for AVIA but definitely not for mid360. Checksum should be added and global time should be "hhmmss.00" format instead of second counts.

leoxiang66, 2025-02-17:
> Anyone who came from https://github.com/sheng00125/LIV_handhold or https://gitee.com/gwmunan/fast-livo_-reproduction trying to set hardware trigger might face the same problem. The original GPRMC [...]

您好， 请问mid360是否不支持PPS秒脉冲触发？