Livox-SDK/Livox-SDK2 #80: No source of built-in IMU to LiDAR extrinsics

Marselka, 2024-06-03:
Could someone prompt where lidar to built-in IMU extrinsics are usually obtained from? I use mid 360, for instance.

User Manual - no info
Livox Viewer 2 - no extrinsics
LiDAR Information Push UDP packets do not have the field with transformation

JokerYan, 2024-07-08:
The user menu includes the IMU extrinsics.
[image]

Marselka, 2024-07-12:
@JokerYan Thanks a lot!

Tech support by email provided also the following, if one interested:

The built-in IMU of the LiDAR is for SMT installation, and the relative value of the installation position is fixed; Hexahedral calibration will be performed at the factory, and this calibration process has been completed. Customers do not need additional IMU calibration and can use it directly.

closed 2024-07-12 by Marselka