Livox-SDK/Livox-SDK2 #81: Frame rate issue

hgdly, 2024-06-21:
Hello,

I use the Livox SDK 2 with the Livox HAP Tx. I used the samples available on github to acquire point clouds. But it seems to me that the number of points I receive is much lower than the number of points per second indicated in the specs of the lidar hap. Is this normal? On livox viewer everything works fine. It's just that when I use the sdk, I get about 1000 points per second instead of the 452000 (or 425000 I don't remember) indicated in the specs.
How can I get more data ?

Thanks for your help.

MaxiStocker, 2024-08-02:
Hi,

The same thing is happening for me. I would appreciate it if someone could answer this issue.

Thanks.

MaxiStocker, 2024-08-06:
So, I have been in contact with the Livox support team, they seem to have forgotten about this repo. In their words: "Thank you for your reminder"

I did however get an answer out of them regarding the frame rate issue. This SDK does not use frames, opposed to other LIDARs, that let you process x points at once in a vectorised manner, the Livox SDK processes each point individually. Apperently, even with c code and a powerfull CPU, this results in a MUCH lower sample rate than advertised. 

As far as I am concerned, this makes the SKD and the Livox Hap unusable.

hgdly, 2024-09-10:
Hi @MaxiStocker!
I've come up with a solution: store the result in binary, then make an additional program to translate the binary into text data.
Hope this helps!