Livox-SDK/Livox-SDK2 #90: Error in the command of cmake .. && make -j

hawk-acc, 2024-12-23:
Hi all,

I encountered the issue after executing sudo make install in the process of installing Livox-SDK2 in Ubuntu 24.04.

Here is the following errors I've been getting:
[  1%] Building CXX object sdk_core/CMakeFiles/livox_lidar_sdk_static.dir/device_manager.cpp.o
In file included from /home/hawk-cheong/Livox-SDK2/sdk_core/../3rdparty/spdlog/spdlog/fmt/fmt.h:21,
                 from /home/hawk-cheong/Livox-SDK2/sdk_core/../3rdparty/spdlog/spdlog/common.h:28,
                 from /home/hawk-cheong/Livox-SDK2/sdk_core/../3rdparty/spdlog/spdlog/spdlog.h:12,
                 from /home/hawk-cheong/Livox-SDK2/sdk_core/base/logging.h:27,
                 from /home/hawk-cheong/Livox-SDK2/sdk_core/device_manager.cpp:36:
/home/hawk-cheong/Livox-SDK2/sdk_core/../3rdparty/spdlog/spdlog/fmt/bundled/format.h:407:6: warning: identifier ‘char8_t’ is a keyword in C++20 [-Wc++20-compat]
  407 | enum char8_t: unsigned char {};
      |      ^~~~~~~
In file included from /home/hawk-cheong/Livox-SDK2/sdk_core/device_manager.h:30,
                 from /home/hawk-cheong/Livox-SDK2/sdk_core/device_manager.cpp:30:
/home/hawk-cheong/Livox-SDK2/sdk_core/comm/define.h:235:8: error: ‘uint8_t’ in namespace ‘std’ does not name a type; did you mean ‘wint_t’?
  235 |   std::uint8_t  dev_type;
      |        ^~~~~~~
      |        wint_t
/home/hawk-cheong/Livox-SDK2/sdk_core/comm/define.h:237:8: error: ‘uint16_t’ in namespace ‘std’ does not name a type; did you mean ‘wint_t’?
  237 |   std::uint16_t cmd_port;
      |        ^~~~~~~~
      |        wint_t
/home/hawk-cheong/Livox-SDK2/sdk_core/comm/define.h:342:8: warning: elaborated-type-specifier for a scoped enum must not use the ‘class’ keyword
  342 |   enum class SyncTimeType : std::uint8_t {
      |   ~~~~ ^~~~~
      |        -----
/home/hawk-cheong/Livox-SDK2/sdk_core/comm/define.h:342:14: error: use of enum ‘SyncTimeType’ without previous declaration
  342 |   enum class SyncTimeType : std::uint8_t {
      |              ^~~~~~~~~~~~
/home/hawk-cheong/Livox-SDK2/sdk_core/comm/define.h:342:34: error: ‘uint8_t’ is not a member of ‘std’; did you mean ‘wint_t’?
  342 |   enum class SyncTimeType : std::uint8_t {
      |                                  ^~~~~~~
      |                                  wint_t
/home/hawk-cheong/Livox-SDK2/sdk_core/comm/define.h:342:42: error: default member initializer for unnamed bit-field
  342 |   enum class SyncTimeType : std::uint8_t {
      |                                          ^
/home/hawk-cheong/Livox-SDK2/sdk_core/comm/define.h:344:3: error: expected ‘;’ at end of member declaration
  344 |   } type;
      |   ^
      |    ;
/home/hawk-cheong/Livox-SDK2/sdk_core/comm/define.h:344:5: error: ‘type’ does not name a type; did you mean ‘typeof’?
  344 |   } type;
      |     ^~~~
      |     typeof
make[2]: *** [sdk_core/CMakeFiles/livox_lidar_sdk_static.dir/build.make:90: sdk_core/CMakeFiles/livox_lidar_sdk_static.dir/device_manager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:223: sdk_core/CMakeFiles/livox_lidar_sdk_static.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

Could someone help me out?

Jquarker, 2025-01-21:
这边遇到了同样的问题

lukeliao, 2025-02-25:
here is my diff:
```
diff --git a/sdk_core/comm/define.h b/sdk_core/comm/define.h
index 66996bf..a406072 100644
--- a/sdk_core/comm/define.h
+++ b/sdk_core/comm/define.h
@@ -26,6 +26,7 @@
 #define LIVOX_DEFINE_H_

 #include <stdio.h>
+#include <cstdint>
 #include <string>
 #include <memory>
 #include <functional>
diff --git a/sdk_core/logger_handler/file_manager.h b/sdk_core/logger_handler/file_manager.h
index 242a431..d8a564a 100644
--- a/sdk_core/logger_handler/file_manager.h
+++ b/sdk_core/logger_handler/file_manager.h
@@ -25,6 +25,7 @@
 #ifndef LIVOX_FILE_MANAGER_
 #define LIVOX_FILE_MANAGER_

+#include <cstdint>
 #include <string>
 #include <vector>
 #include <map>
```
it's work for me on ubuntu 24.04, gcc 13.3.
later I will get a pull-request for this.

PaulKupka, 2025-04-04:
> here is my diff:
>
> ```
> diff --git a/sdk_core/comm/define.h b/sdk_core/comm/define.h
> index 66996bf..a406072 100644
> --- a/sdk_core/comm/define.h
> +++ b/sdk_core/comm/define.h
> @@ -26,6 +26,7 @@
>  #define [...]

could you explain how to use that code you posted?
what do I do with it?

oskarekxy, 2025-04-08:
@PaulKupka If you still have not resolved your issue, you should modify files according to @NaokiTakahashi12 PR.

[image]

AlZi-88, 2025-05-17:
Also tested it and works fine in ubuntu 24.04 and g++ 13.3

Jerry1962325, 2025-07-06:
> [@PaulKupka](https://github.com/PaulKupka) If you still have not resolved your issue, you should modify files according to [@NaokiTakahashi12](https://github.com/NaokiTakahashi12) PR.如果您仍然没有解决您的问题，您应该根据 PR 修改文件。
> 
> [image]

Great thanks！